ve3wwg/stm32f103c8t6 #2: Missing step in README?

beckdac, 2017-04-16:
Based on #5 as an addition to the `README.md`
https://github.com/beckdac/stm32f103c8t6/commit/5664d6b663a4f1add2aa364b4fc23c34d4473115#diff-04c6e90faac2675aa89e2176d2eec7d8

Sorry I don't have a clean pull request for this.  I have tried to match the narrative style.

ve3wwg, 2017-06-01:
The source code in this project is still evolving. However, as it pertains to libwwg, you should now find that there are Makefile(s) at the levels to build what you need.

ve3wwg, 2018-04-06:
It is possible that you need to do the top level make first, which should have that covered (from the top it should compile everything). I can see how your experience might happen if you build in piece meal initially. The repo should be stable now with the exception that not all subdirs containing projects are covered in the book. The ones that are not covered by the book may be in various states of repair.

Yes, miniblink doesn't require the library, but is likely tied to it by the project Makefile.

Thanks for your comments, especially the all opened source aspect. I am very much "done" with Microsoft these days (life at home has improved without MS as well, with my wife and mother using MacOS now). With proprietary IDEs you're always futzing around to make the old software work with the new version/releases etc. Online sites go dark so tying things to online IDEs/resources is even worse in some regards. Not to mention licensing terms which can expire or restrict. So yes, I believe in open source and believe there is an advantage to people like you investing time in writing software.

I hope you eventually get a copy of that book- it should be out in September. The first draft, tech review and 2nd draft are completed. But through the summer a bunch of other folks at Apress will be working hard to get it ready for printing/ebook. I've basically invested a year of research and writing effort already to get to this point. But it's been a fun ride and I very much like the stm32.

closed 2018-05-20 by ve3wwg