ve3wwg/stm32f103c8t6 #4: Can't compile rtos/libwwg.

Tolmeros, 2018-05-19:
```
arm-none-eabi-gcc -Os -g -std=c99 -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -I/stm32/stm32f103c8t6_freertos/libopencm3/include -I/stm32/stm32f103c8t6_freertos//rtos/libwwg/include -I./rtos -I.  -MD -Wall -Wundef -DSTM32F1 -I/stm32/stm32f103c8t6_freertos/libopencm3/include -I/stm32/stm32f103c8t6_freertos//rtos/libwwg/include  -o getline.o -c getline.c
getline.c:5:20: fatal error: memory.h: No such file or directory
#include <memory.h>
                    ^
compilation terminated.
```

Ubuntu 16.04 amd64
gcc-arm-none-eabi/xenial,now 15:4.9.3+svn231177-1

ve3wwg, 2018-05-20:
Hi:

The problem is with your cross compiler. On my system (as I have it installed), the missing header file is found at:

    /opt/gcc-arm/arm-none-eabi/include/memory.h

So your cross compiler should have this file.  It is also possible that it exists but the compiler is not searching it by default. Try adding a -I/whatever/include to the TGT_CFLAGS macro in file Makefile.incl at the top level.

Tolmeros, 2018-05-23:
I tried to understand the reason for this problem.
`memory.h` is a part of `libnewlib-arm-none-eabi`, but it was provided in 2.5.0 version.
( [https://sourceware.org/git/gitweb.cgi?p=newlib-cygwin.git;a=commit;h=9fbd510569df2916cbcc69d46e1d0da9f63643e1](https://sourceware.org/git/gitweb.cgi?p=newlib-cygwin.git;a=commit;h=9fbd510569df2916cbcc69d46e1d0da9f63643e1) )

The version of the `libnewlib-arm-none-eabi` are `2.2.0+git20150830.5a3d536-1` at the official Ubuntu repositories.
It seems slightly out of date. 

But, when I sought the essence of the problem, I found an solutions like this:

1) [https://gnunet.org/bugs/view.php?id=2511](https://gnunet.org/bugs/view.php?id=2511)

```
#if HAVE_MEMORY_H
#include <memory.h>
#else
#include <string.h>
#endif
```

2) [http://crossgcc.s0ufcgwqre.narkive.com/1ARzsDSY/patch-00-of-14-ct-ng-fixes-for-mingw-target-and-host-compilation-with-samples#post9](http://crossgcc.s0ufcgwqre.narkive.com/1ARzsDSY/patch-00-of-14-ct-ng-fixes-for-mingw-target-and-host-compilation-with-samples#post9)

```
#ifdef HAVE_STRING_H
# if !defined STDC_HEADERS && defined HAVE_MEMORY_H
# include <memory.h>
# endif
# include <string.h>
#endif
```

I tried the first variant with `rtos/libwwg/src/getline.c` and toolchain from official repositories.
Now it is compiling and running well.

ve3wwg, 2018-05-23:
This unfortunately, doesn't work for everyone. Your cross tools has HAVE_MEMORY_H defined, where other builds will not have this macro. All I can do is put a comment in there for now.

closed 2018-09-24 by ve3wwg

fbmnds, 2018-12-14:
There should be a remark in the Prerequisites section of the Readme.

sharma-kullu, 2019-03-24:
```
`$ make
arm-none-eabi-gcc -Os -g -std=c99 -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -I/home/alam/stm32f3/libopencm3/include -I/home/alam/stm32f3//rtos/libwwg/include -I./rtos -I.  -MD -Wall -Wundef -DSTM32F1 -I/home/alam/stm32f3/libopencm3/include -I/home/alam/stm32f3//rtos/libwwg/include  -o usbcdc.o -c usbcdc.c
usbcdc.c:7:10: fatal error: libopencm3/cm3/scb.h: No such file or directory
 #include <libopencm3/cm3/scb.h>
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [../../../Makefile.incl:126: usbcdc.o] Error 1

```
`Is there some problem with newer libopencm3 versions or am I missing something in Makefile.incl ?
libopencm3 compiles well so does installation of gnu toolchain
Cygwin for windows

ve3wwg, 2019-03-24:
> `$ make
> arm-none-eabi-gcc -Os -g -std=c99 -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes [...]

Is it possible that you didn't download and populate libopencm3 in the
correct subdirectory? Or is it possible that you pulled down the source
code while they were making changes. Try:

$ cd ~/stm32f103c8t6/libopencm3
$ git pull

to bring down any changes/fixes.

When I check github, I see that the file still exists there. I suggest that
if the above does not correct the issue, that you delete the libopenfm3
subdirectory and recreate it. The Makefile is designed to see it in
~/stm32f103c8t6/libopencm3 subdirectory. If you placed it somewhere else
then several Makefile changes will be required. It is easiest for you to
recreate the structure that I used in the book.

Warren

sharma-kullu, 2019-03-25:
Well in the Makefile.incl in the top directory I changed the TOP_DIR, OPENCM3_DIR to absolute paths
Make begin to work

ve3wwg, 2019-03-25:
Yes, it needs to be an absolute path.

byunghwee, 2019-06-16:
Well in the Makefile.incl in the top directory I changed the TOP_DIR, OPENCM3_DIR to absolute paths
Make begin to work

I have a same problem.
Could you let me know how you fix it?

TOP_DIR 	:= $(dir $(abspath $(lastword $(MAKEFILE_LIST))))
OPENCM3_DIR 	:= $(abspath $(dir $(abspath $(lastword $(MAKEFILE_LIST))))/libopencm3)

==> How to change?

ve3wwg, 2019-06-17:
The abspath function needs GNU make. You may have it installed as gmake. Otherwise just set the variables directly:

TOP_DIR = "path1"
OPENCM3_DIR = "path2"

Gus-Ghedim, 2019-11-19:
I am also having the same issue:

`make -C rtos/libwwg
make[1]: Entering directory '/cygdrive/c/Users/arge10/Documents/stm32/stm32f1/rtos/libwwg'
make -w -C ./src
make[2]: Entering directory '/cygdrive/c/Users/arge10/Documents/stm32/stm32f1/rtos/libwwg/src'
File not found - *.[ch]
arm-none-eabi-gcc -Os -g -std=c99 -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -I/include -I/rtos/libwwg/include -I./rtos -I.  -MD -Wall -Wundef -DSTM32F1 -I/include -I/rtos/libwwg/include  -o usbcdc.o -c usbcdc.c
usbcdc.c:7:10: fatal error: libopencm3/cm3/scb.h: No such file or directory
    7 | #include <libopencm3/cm3/scb.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [../../../Makefile.incl:111: usbcdc.o] Error 1
make[2]: Leaving directory '/cygdrive/c/Users/arge10/Documents/stm32/stm32f1/rtos/libwwg/src'
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/cygdrive/c/Users/arge10/Documents/stm32/stm32f1/rtos/libwwg'
make: *** [Makefile:41: libwwg] Error 2`

I checked the path of the files and all they are present in the correct ones, I tried also first using Cygwin, then Mingw and now again Cygwin. However, the same issue still persists.

Someone that faced this problem and could solve it could show me the way?

ve3wwg, 2019-11-20:
If you're not running Linux/MacOS/*BSD, then I believe the best recipe is one of the following:

1. [Linux for Windows](https://docs.microsoft.com/en-us/windows/wsl/install-win10)
1. Linux under [VirtualBox](https://www.virtualbox.org/) on your system of choice

It may be possible to get the environment up on Cygwin/MingW, but it requires a lot of expert tweaking, much more than I am prepared to do.

Gus-Ghedim, 2019-11-21:
I tried with Linux for Windows and it worked perfectly

sharma-kullu, 2019-11-21:
Yes! The issue remains with posix environments running on windows.Its
better to do it in Linux native environment. I found issues while using
Libusb on cygwin as well.

Ex- Researcher
CSIR-CSIO
Delhi Center

chilltk, 2024-01-05:
hi guys, need help with this issue, windows 10, cygwin. all was smooth until build. is it any solution?
```
make -C libopencm3 TARGETS=stm32/f1
  GENHDR  stm32/f1
  BUILD   lib/stm32/f1
  CC      adc.c
  CC      adc_common_v1.c
  CC      can.c
  CC      crc_common_all.c
  CC      dac_common_all.c
  CC      dac_common_v1.c
  CC      desig_common_all.c
  CC      desig_common_v1.c
  CC      dma_common_l1f013.c
  CC      exti_common_all.c
  CC      flash.c
  CC      flash_common_all.c
  CC      flash_common_f.c
  CC      flash_common_f01.c
  CC      gpio.c
  CC      gpio_common_all.c
  CC      i2c_common_v1.c
  CC      iwdg_common_all.c
  CC      pwr_common_v1.c
  CC      rcc.c
  CC      rcc_common_all.c
  CC      rtc.c
  CC      spi_common_all.c
  CC      spi_common_v1.c
  CC      timer.c
  CC      timer_common_all.c
  CC      usart_common_all.c
  CC      usart_common_f124.c
  CC      mac.c
  CC      mac_stm32fxx7.c
  CC      phy.c
  CC      phy_ksz80x1.c
  CC      usb.c
  CC      usb_control.c
  CC      usb_standard.c
  CC      usb_msc.c
  CC      usb_hid.c
  CC      usb_audio.c
  CC      usb_cdc.c
  CC      usb_midi.c
  CC      usb_dwc_common.c
  CC      usb_f107.c
  CC      st_usbfs_core.c
  CC      st_usbfs_v1.c
  CC      vector.c
  CC      systick.c
  CC      scb.c
  CC      nvic.c
  CC      assert.c
  CC      sync.c
  CC      dwt.c
  AR      libopencm3_stm32f1.a
make -C rtos/libwwg
make[1]: Entering directory '/cygdrive/c/Users/timur/Documents/repos/stm32f103c8t6/rtos/libwwg'
make - -C ./src
make[2]: Entering directory '/cygdrive/c/Users/timur/Documents/repos/stm32f103c8t6/rtos/libwwg/src'
File not found - *.[ch]
../../../Makefile.incl:105: *** multiple target patterns.  Stop.
make[2]: Leaving directory '/cygdrive/c/Users/timur/Documents/repos/stm32f103c8t6/rtos/libwwg/src'
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/cygdrive/c/Users/timur/Documents/repos/stm32f103c8t6/rtos/libwwg'
make: *** [Makefile:41: libwwg] Error 2
```

sharma-kullu2, 2024-01-05:
Just my POV, I ll suggest to better move to latest STM32Cubef1 firmware, either use it with stm ide or 3rd party ide like platformio. Compilation, building and uploading is much better and time consuming, even for beginners.

ve3wwg, 2024-01-05:
Two things:

   1. The make/gmake command assumes you will use the bash shell.
   2. I now recommend that you use WSL2 under Windows.

Notes:

   1. The libopencm3 library has made some changes that will cause warnings
   and perhaps a couple of errors in some code. There is no way to anticipate
   this in the writing of the book.
   2. A 2nd edition is coming out in the spring that will provide detailed
   WSL2 instructions for Windows users and include the
   necessary libopenm3 changes.

Thanks, Warren