ve3wwg/stm32f103c8t6 #14: rtos/uart2 example not working

VictorLamoine, 2019-03-17:
I have successfully ran the `rtos/uart` example but the `rtors/uart2` example does not work for me.

Hardware:
- Blue Pill
- FT232RL breakout board
- ST-Link V2

Software:
- Kubuntu 18.04
- FreeRTOS 10.0.1 and 10.2.0 (I have tested both)
- arm-none-eabi-gcc 6.3.1 

# Chapter 6 - Project uart2
The example compiles and uploads successfully however I get no output in the serial terminal.

I have tried a few simple modifications to identify the problem.

In the main after configuring the GPIO I switch off the PC13 LED:
```cpp
  gpio_set(GPIOC, GPIO13);
```

I blink the PC13 LED in the `demo_task`
```bash
static void
demo_task(void *args __attribute__((unused))) {

	for (;;) {
    gpio_toggle(GPIOC,GPIO13);
		uart_puts("Now this is a message..\n\r");
		uart_puts("  sent via FreeRTOS queues.\n\n\r");
		vTaskDelay(pdMS_TO_TICKS(1000));
	}
}
```

The LED switches one once but never blinks. The `uart_puts` call is blocking and never seems to exit.

Do you have any idea how to solve the problem?

VictorLamoine, 2019-03-17:
I have added a check for the queue creation, in case it was failing:
```cpp
  if (uart_txq == 0)
  {
    for (;;) {
      gpio_toggle(GPIOC,GPIO13);
      for (uint32_t i = 0; i < 80000; i++)
        __asm__("nop");
    };
  }
```

It does not seem to fail as the LED does not blink (with just this modification).

VictorLamoine, 2019-03-18:
I tried a simpler example with no success:
```bash
#include <FreeRTOS.h>
#include <libopencm3/stm32/gpio.h>
#include <libopencm3/stm32/rcc.h>
#include <queue.h>
#include <task.h>

static QueueHandle_t queue;

static void
blink_task(void *args __attribute__((unused)))
{
  bool nothing;
  while (1)
  {
    if (xQueueReceive(queue, &nothing, 10) == pdPASS)
      gpio_set(GPIOC, GPIO13); // Turn off
    else
      taskYIELD(); // Yeld so that queue_task can run
  }
}

static void
queue_task(void *args __attribute__((unused)))
{
  bool nothing = false;
  while (1)
  {
    gpio_clear(GPIOC, GPIO13); // Turn on
    vTaskDelay(pdMS_TO_TICKS(100));
    xQueueSend(queue, &nothing, portMAX_DELAY);
    vTaskDelay(pdMS_TO_TICKS(1000));
  }
}

int
main(void)
{
  rcc_clock_setup_in_hse_8mhz_out_72mhz();

  // Blue-Pill led
  rcc_periph_clock_enable(RCC_GPIOC);
  gpio_set_mode(
    GPIOC,
    GPIO_MODE_OUTPUT_2_MHZ,
    GPIO_CNF_OUTPUT_PUSHPULL,
    GPIO13);
  gpio_set(GPIOC, GPIO13); // Turn off

  queue = xQueueCreate(32, sizeof(bool));
  xTaskCreate(blink_task, "BLINK", 100, NULL, configMAX_PRIORITIES-1, NULL);
  xTaskCreate(queue_task, "QUEUE", 100, NULL, configMAX_PRIORITIES-2, NULL);
  vTaskStartScheduler();
  while(1);
  return 0;
}
```

Expected behavior:
- `main` 
  - Configures the clocks
  - Configures GPIO for Blue Pill LED
  - Turns off the led
  - Creates a queue and schedule two tasks.
  - Runs the scheduler

- `task_send`
  - Turn on the LED
  - Wait 100 ms
  - Push a message in the queue (content does not matter here)
  -  Wait 1 sec

`task_receive`
  - Check if a message is in the queue
    - Yes: turn off the led
    - No: yeld

I expect the led to be turned on for 100 ms and then turned off for 900 ms.

The led is always on, the execution of the program is blocking at `xQueueSend`.

ve3wwg, 2019-03-19:
Have you tried a different pcb? This has been tested by myself and others and does work.  It is possible that the receiving pin of the peripheral has been damaged by ESD.

VictorLamoine, 2019-03-19:
Thanks for the insight!

I've just tried three others PCBs and none work with the UART2 example. They all work with the UART example. I've also asked the same question [here](https://stackoverflow.com/questions/55217781/why-is-the-xqueuesendtoback-call-blocking).

If you have some suggestions I'm all ears, right now I'm stuck at page 85 of your book :sob:

ve3wwg, 2019-03-21:
I've pushed up two minor fixes to uart.c in the uart2 project.

1. The comments at the top of the program erroneously had RX and TX pins reversed.
1. Modified the tasks to use the _same_ priority. The original code got away with this only because of the delay.

It is likely you wired the serial port according to the top of uart.c (this was incorrect). Put RX on A9 and TX on A10 (like it was for the first uart project), and your project should work.

If you are using git, do a "git pull" to pull down the updates.

VictorLamoine, 2019-03-21:
The UART example was already working so I got it plugged the right way.
Modifying the task priority does not help, the problem seems to be deeper, I have asked the same 
question on the [FreeRTOS discussion forum](https://sourceforge.net/p/freertos/discussion/382005/thread/8127550229/)

ve3wwg, 2019-03-21:
Did you read "The comments at the top of the program erroneously had RX and TX pins reversed."?  If you reattached the wires after trying the first example, you may have miswired it according to the program comments.

I was able to run the example just fine last night (FreeRTOS 10.0.1 ). It worked with the task priority the same or as originally coded (but using the same priority is recommended).

Have you retested minicom with the TTL UART looped so that RX connected to TX and typed for echoed characters? Make doubly sure that your TTL UART and minicom is _still_ working. Make sure that _hardware flow control is disabled_, as the project requires. In these matters, every little detail matters.

VictorLamoine, 2019-03-21:
Yes I have doubled checked cabling and I have tested both `minicom` and `screen`.

If I run a `gdb` session on the `rtos/uart2` program the `xQueueSend` is a blocking call that never exits  and thus prevents everything else from running (including the RTOS scheduler).

The problem I have does not lay in the UART (I can remove the UART and still reproduce the problem) but with FreeRTOS / libopencm3 combination. I have tested both FreeRTOS 10.0.1 and 10.2.0 with no change.

In every test that I have ran `xQueueSend` is a blocking call that never exits and I fail to find why. I'll keep you updated if I find the error, if you have other suggestions to test I'll try anything  :man_scientist: :microscope:

VictorLamoine, 2019-03-21:
Kubuntu 18.04 ships with `arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620`, with this compiler the code does not work.

`memcpy` seems to be the problematic function call in the code, it is called by FreeRTOS when adding an element to the queue. (details in the FreeRTOS discussion on SourceForge)

If I use the `Version 8-2018-q4-major Linux 64-bit` compiler then the code executes fine. It can be downloaded here: https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads.

Thank you for your help!

closed 2019-03-21 by VictorLamoine

ve3wwg, 2019-03-21:
Ok, so this is chalked up to another compiler bug affecting FreeRTOS (in this case). I will add a note to this effect in the README, to avoid others from struggling with this.