ve3wwg/stm32f103c8t6 #22: Build Error: stm32f103c8t6\miniblink with Makefile "No rule to make target 'miniblink.elf'

ElusiveTau, 2020-01-13:
I'm using GNU `make` which I installed using `MinGW32` (MinGW Installation Manager). About the `make` utility I am using:
> GNU Make 4.0
> Built for Windows32
> Copyright (C) 1988-2013 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.

When I run the make command inside `\stm32f103c8t6\miniblink` I get the `File not Found` error:  

> C:\Users\Minh Tran\Desktop\Code\BeginningSTM32WithFreertos\stm32f103c8t6\miniblink>make
> File not found - *.[ch]
> make: *** No rule to make target 'miniblink.elf', needed by 'elf'.  Stop.

fivdi, 2020-01-26:
@ElusiveTau Were all steps mentioned in the [PREREQUISITES section of the README](https://github.com/ve3wwg/stm32f103c8t6#prerequisites) successfully completed before attempting to make the miniblink example?

ElusiveTau, 2020-01-27:
Yes.